maxgerhardt/rotary-encoder-over-mcp23017 #1: How to handle buttons

Daniele71, 2019-01-26:
Hello, first of all, thanks for sharing your work !
I 've the polling example running with 1 encoder on a Teensy 3.2 board, so far so good.
How about buttons ?
With 16 encoders, I would use 2 mcp for encoders and 1 for buttons but how exactly ?
I think that the third mcp should be handled in different way...

Honestly, I find the documentation a bit too "essential"; a more "real life" example would help a lot.

Thanks,
Daniele.

maxgerhardt, 2019-02-04:
Hi @Daniele71, I will look into these examples later this week if I have the time.

ssyomik, 2019-02-17:
Hi, @Daniele71 !
Here is my simple example, how to use several MCPs with several encoders and buttons. This way you can use encoders and (their) buttons on same or different MCPs. This is not the best and most effective way, but it works.

This way you can click, rotate and push+rotate. Anyway, there is room for further development.

I used a [Polling](https://github.com/maxgerhardt/rotary-encoder-over-mcp23017/tree/master/examples/Polling) example and [Bounce2MCP](https://github.com/cosmikwolf/Bounce2mcp) fork from @cosmikwolf. 

```c++
#include <Bounce2mcp.h>
#include <Arduino.h>
#include <Wire.h>
#include <Adafruit_MCP23017.h>
#include <Rotary.h>
#include <RotaryEncOverMCP.h>

#define NUM_BUTTONS 4
// Total amount of buttons 
const uint8_t BUTTON_PINS[NUM_BUTTONS] = {8, 8, 8, 8};
// Pins, used by buttons (each on "their" MCP)
bool BUTTON_ROTATION[NUM_BUTTONS] = {false, false, false, false};
// Flags to distinguish click from turn

BounceMcp * buttons = new BounceMcp[NUM_BUTTONS];
// Button debounce wrappers

Adafruit_MCP23017 mcp1, mcp2, mcp3, mcp4;
// Initialize all MCPs

Adafruit_MCP23017* allMCPs[] = { &mcp1, &mcp2, &mcp3, &mcp4 };
constexpr int numMCPs = (int)(sizeof(allMCPs) / sizeof(*allMCPs));

Adafruit_MCP23017* BUTTON_MCPS[NUM_BUTTONS] = {&mcp1, &mcp2, &mcp3, &mcp4};
// Pointers to MCPs to which each button is attached, respectively

void RotaryEncoderChanged(bool clockwise, int id);

RotaryEncOverMCP rotaryEncoders[] = {
        RotaryEncOverMCP(&mcp1, 0, 1, &RotaryEncoderChanged, 0),
        RotaryEncOverMCP(&mcp2, 0, 1, &RotaryEncoderChanged, 1),
        RotaryEncOverMCP(&mcp3, 0, 1, &RotaryEncoderChanged, 2),
        RotaryEncOverMCP(&mcp4, 0, 1, &RotaryEncoderChanged, 3)
};
constexpr int numEncoders = (int)(sizeof(rotaryEncoders) / sizeof(*rotaryEncoders));

void RotaryEncoderChanged(bool clockwise, int id) {
    buttons[id].update();
    Serial.println("Encoder " + String(id) + ": "
            + (clockwise ? String("clockwise ") : String("counter-clock-wise "))
            + (buttons[id].read() ? String("") : String("(pressed)")));
    BUTTON_ROTATION[id] = true;
    // Change the rotation flag
}

void setup(){
    Serial.begin(115200);
    Serial.println("MCP23007 Polling Test");

    mcp1.begin(0); 
    mcp2.begin(1); 
    mcp3.begin(2); 
    mcp4.begin(4); 
    // Simple wiring A0-A2 to 3.3V

    for (int i = 0; i < NUM_BUTTONS; i++) {
      // Setting up buttons on their MCPs
      // According to https://github.com/cosmikwolf/Bounce2mcp
      BUTTON_MCPS[i]->pinMode(BUTTON_PINS[i], INPUT);
      BUTTON_MCPS[i]->pullUp(BUTTON_PINS[i], HIGH);
      // My HW-040 (KY-040) modules have no pullups, so turning it on
      buttons[i].attach(*BUTTON_MCPS[i], BUTTON_PINS[i], 5);
    }

    for(int i=0; i < numEncoders; i++) {
        rotaryEncoders[i].init();
    }
}

void pollAll() {
    for(int j = 0; j < numMCPs; j++) {
        uint16_t gpioAB = allMCPs[j]->readGPIOAB();
        for (int i=0; i < numEncoders; i++) {
            if(rotaryEncoders[i].getMCP() == allMCPs[j])
                rotaryEncoders[i].feedInput(gpioAB);
        }
    }
}

void loop() {
    pollAll();

  for (int i = 0; i < NUM_BUTTONS; i++)  {
    buttons[i].update();
    // Updating all debouncers
    if ( buttons[i].fell() ) {
      //If button was pressed down...
//      Serial.println("Button "+String(i)+":  down");
      BUTTON_ROTATION[i] = false;
      // Clearing the rotation flag
    }
    
    if ( buttons[i].rose() ) {
      // When button was released...
//      Serial.println("Button "+String(i)+": up");
      if (BUTTON_ROTATION[i]) {
        // If there was any rotation of that encoder
//        Serial.println("No, it does not count!");
      }
      else
      {
        // If there was no rotation
        Serial.println("Button "+String(i)+": click");
      }
    }
  }
}
```

Daniele71, 2019-02-17:
Thanks ! I will try your example in the next days.

Daniele71, 2019-02-26:
> Hi, @Daniele71 !
> Here is my simple example, how to use several MCPs with several encoders and buttons. This way you can use encoders and (their) buttons on same or different MCPs. This is not the [...]

Hi, very busy now so I did nothing with this new code..
After only a quick look, there's something that does not convince me too much. You did a great job using readGPIOAB() instead of digitalRead() on each pin. Now with Bounce2mcp we have both...
So, what you did is partly lost. Great flexibility but more traffic on i2c bus.
What do you think ?

lbenelli, 2020-05-06:
Hi everyone, 

First off, thanks for all the code already written. 

I got the interrupt version running with multiple encoders, I am now trying to add the buttons. The example above seems to use one button on each mcp if i got this right. 

Has anyone an example of rotary enocders AND buttons on a single MCP or care to explain how i should proceed? 

Still learning... 

Thanks, Luca

maxgerhardt, 2021-02-07:
Closing issue due to this project only being about rotary encoders and a given example sketch that actually uses buttons above.

closed 2021-02-07 by maxgerhardt