maxgerhardt/rotary-encoder-over-mcp23017 #2: Speed or velocity its posible

jmym, 2019-03-05:
Hi, i use in my project  4 mcps and how i implement a velocity in the encoders, they read very slow. I need send MIDI message values from 0 to 127.

Could you help me, please?

maxgerhardt, 2019-03-05:
> Hi, i use in my project 4 mcps and how i implement a velocity in the encoders, they read very slow
What time between the readouts do you have? 

>I need send MIDI message values from 0 to 127.

What does this have to do with the first sentence? The problem is just that the rotary encoders aren't read fast enough, right?

jmym, 2019-03-05:
Yes, the steps between 1 to 20 for example, the encoder its moved full(i
need more movements to complete the 127). And i need to move full, from 0
to 127, only one time.

maxgerhardt, 2019-03-05:
I'm not sure if I really understand what you want: So you e.g. wait for 0.5 seconds and see how much the rotary encoder was moved in that time and send that as "velocity"? Or do you just want to map total left -> 0 and total right -> 20? Because that's actually hard because you don't know the absolute position of the encoder, only movements.

jmym, 2019-03-06:
I have this library for a single encoder.

https://github.com/craftyc0der/arduino-rotary-encoder-with-velocity/blob/master/RotaryEncoder.cpp

And this one for logarithmic velocity.

https://github.com/MajicDesigns/MD_REncoder/blob/master/src/MD_REncoder.cpp

Sorry for my horrible english.

Regards,
JM

maxgerhardt, 2019-03-06:
Okay I kind of get what you might want to do. 

The code at 

https://github.com/MajicDesigns/MD_REncoder/blob/c94e8f11697631d93960bacea217d57a04c27853/src/MD_REncoder.cpp#L103-L112

together with 

https://github.com/MajicDesigns/MD_REncoder/blob/master/src/MD_REncoder.h#L154-L165

counts the number of rotations in the given time period. Since my library already has a callback function for whenever a rotation occurs, you can do the exact same type of computations in the callback function (i.e., `RotarayEncoderChanged()`)

closed 2019-03-06 by maxgerhardt