maxgerhardt/rotary-encoder-over-mcp23017 #3: Having trouble adding the library to the arduino 1.8.8 IDE

maartencz, 2019-03-06:
Hi,

Am a newbie in Arduino. I have found your rotary-encoder-over-mcp23017 library, but having trouble compiling. I get the error added between double dashed lines below.

I did download the zip files containing the rotary and Adafruit-MCP23017-Arduino-Library files. They could be added by in the Arduino 1.8.8 IDE, by navigateing to Sketch > Include Library > Add .ZIP Library. At the top of the drop down list, select the option to "Add .ZIP Library''.

However I have been unable to add the rotary-encoder-over-mcp23017 library in the similar way.
Tried to copy some files into a structure that looks like the other two zip libraries, but I guess I am missing the RotaryEncOverMCP.cpp file.

It would be nice of you to add your own library as a downloadable zip file next to the rotary and Adafruit-MCP23017-Arduino-Library files.

For sure I lack some knowledge somewhere, and hope you understand my issue and can come with a solution.

Thanks!

============================================================
Arduino: 1.8.9 Hourly Build 2019/02/04 10:33 (Windows 10), Board: "Arduino Leonardo"

rotary_encoder_mcp23017_int:5:30: error: RotaryEncOverMCP.h: No such file or directory

compilation terminated.

exit status 1
RotaryEncOverMCP.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
==========================================================

maxgerhardt, 2019-03-06:
The project is meant to compile with PlatformIO (hence the `platformio.ini`) but sure you can get it working under Arduino IDE, too. 

Go put the https://github.com/maxgerhardt/rotary-encoder-over-mcp23017/blob/master/src/RotaryEncOverMCP.h into a ZIP file and add it as a library. 

Take the two folders from https://github.com/maxgerhardt/rotary-encoder-over-mcp23017/tree/master/lib put them into one zip file each and add them as a library. But make sure you have previously removed other versions of this library. 

Then you be able to take the interrupt or polling example and they should be compilable.

maxgerhardt, 2019-03-06:
Also there is no `RotaryEncOverMCP.cpp` file the library consists only of a header file which includes all necessary functions :P

maxgerhardt, 2019-08-24:
Reopen if you need further help, otherwise I assume this is solved.

closed 2019-08-24 by maxgerhardt