maxgerhardt/rotary-encoder-over-mcp23017 #4: How to use encoders on the PortB side

berbergh, 2019-08-24:
Hi,
I am following your project with great interest and I succeeded to get two encoders connected on the portA side.
My question now is, how to connect encoders to the port B side?

maxgerhardt, 2019-08-24:
As can be seen from the code 

https://github.com/maxgerhardt/rotary-encoder-over-mcp23017/blob/master/src/RotaryEncOverMCP.h#L47

We read out the 16-bit register `AB` and select the bit according to `pinA` and `pinB` number. So if we set `pinA = 0` this wil go to Port A Pin 0, (GPA0), and `pinA = 8` will be Port B pin 0 (GPB0), and so on, until the last possible value of `15`.

berbergh, 2019-08-24:
That is what I thought too, so I canged your InterruptExample, like this

```
RotaryEncOverMCP rotaryEncoders[] = {
        // outputA,B on GPA7,GPA6, register with callback and ID=1
        RotaryEncOverMCP(&mcp, 10, 11, &RotaryEncoderChanged, 1),
        RotaryEncOverMCP(&mcp, 5, 4, &RotaryEncoderChanged, 2),
        RotaryEncOverMCP(&mcp, 9, 8, &RotaryEncoderChanged, 3)
```

When I turn encoder 1 or 2, I get good readings, but turning encoder 3 makes the software hang after 1 interrupt (I added a line to debug):

`void checkInterrupt() {
    if(awakenByInterrupt) {
        // disable interrupts while handling them.
        detachInterrupt(digitalPinToInterrupt(arduinoIntPin));
        Serial.println("int-in");
        handleInterrupt();
        attachInterrupt(digitalPinToInterrupt(arduinoIntPin),intCallBack,FALLING);
    }
}`

maxgerhardt, 2019-08-24:
Does the polling example work?

berbergh, 2019-08-24:
Yes, it does

berbergh, 2019-08-29:
Just for my curiousity: does this project get a follow up?

maxgerhardt, 2019-08-29:
Sorry yeah, I'll try to reproduce your problem over the weekend. Super busy right now.

berbergh, 2019-08-29:
No worries. I was just wondering.

thijstriemstra, 2020-07-23:
> Yes, it does

No, it doesn't, the polling example also crashes when trying to add more encoders.

rote-locke, 2021-02-07:
Hi there,

I used your library to create an input device consisting of six encoders and I am also facing the issue that the device stops working as soon as encoders are added on port B.
Working on the interrupt example I added my encoders as shown below. Encoders 1 to 4 work fine, as all are on port A, but as soon as I add one or more encoders on port B no more inputs are registered. Not even from the encoders on port A previously working.

`RotaryEncOverMCP rotaryEncoders[] = {
        RotaryEncOverMCP(&mcp, 0, 1, &RotaryEncoderChanged, 1),
        RotaryEncOverMCP(&mcp, 2, 3, &RotaryEncoderChanged, 2),
        RotaryEncOverMCP(&mcp, 4, 5, &RotaryEncoderChanged, 3),
        RotaryEncOverMCP(&mcp, 6, 7, &RotaryEncoderChanged, 4),
        RotaryEncOverMCP(&mcp, 8, 9, &RotaryEncoderChanged, 5),
        RotaryEncOverMCP(&mcp, 10, 11, &RotaryEncoderChanged, 6)
};`

Would really appreciate a hint, as this is the one blocking point left on an otherwise completed project.

maxgerhardt, 2021-02-07:
Okay, I'll re-investigate this asap.

rote-locke, 2021-02-07:
> 
> 
> Okay, I'll re-investigate this asap.

Wow, did not expect this fast of a reply. Thanks for having a look!

maxgerhardt, 2021-02-07:
Well as a first investigation, I have rebuild the circuit as shown in the README but I've connected my rotary encoder's CLK/DT (or outputA / outputB) pin to the GPB7 and GPB6 pins. If I then reference these pins as 15 and 14 in the code (GPA0 = 0 ... GPA7 = 7, GPB0 = 8 ... GPB7 = 15), it **does work** in polling mode.

I literally just took the PollingExample.cpp and changed 

```
/* Array of all rotary encoders and their pins */
RotaryEncOverMCP rotaryEncoders[] = {
        // outputA,B on GPB7,GPB6, register with callback and ID=1
        RotaryEncOverMCP(&mcp, 15, 14, &RotaryEncoderChanged, 1)
};
```

and I get movement when I move my rotatary encoder

```
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
MCP23007 Polling Test
Encoder 1: counter-clock-wise
Encoder 1: counter-clock-wise
Encoder 1: clockwise
Encoder 1: clockwise
```

I will test with more rotary encoders to see if that is what's causing the problem, and also the interrupt mode.

rote-locke, 2021-02-07:
Thanks for checking. So far, I've only ever used the interrupt example. I just tested the polling example with my setup of six encoders and got it indeed working right from the start.

I then went back and reloaded the original interrupt example and just added the other encoders, same as before. What I then realised is, that I can use all encoders on port A until I turn one of the encoders on port B. As soon as this happens it seems to crash and even the encoders on port A stop registering.

rote-locke, 2021-02-07:
EDIT: I am running this via Arduino IDE on a Micro Pro clone with one MCP. Goal is to create a controller for software using the [joystick library](https://github.com/MHeironimus/ArduinoJoystickLibrary). Got everything working as it should except the encoders on port B. Made a few of these for different applications aleardy but this is the first time I use a port expander and this many encoders.

maxgerhardt, 2021-02-07:
> I then went back and reloaded the original interrupt example and just added the other encoders, same as before. What I then realised is, that I can use all encoders on port A until I turn one of the [...]

Ok thanks for the hint, I'll try and reproduce exactly that with the interrupt example and one encoder on port A and one on port B.

maxgerhardt, 2021-02-07:
I was able to identify a problem in the InterruptExample that made the application hang. There needs to be a readout of the GPIOAB  register by doing `uint16_t dummy = allMCPs[j]->readGPIOAB();` below the `readINTCAPAB()` call. But this doesn't fix the problem the captured values in there, when an interrupt occurrs, seem to be always 1 for the pins and no 0 is every registered.. Strange. I'll investigate further, with maybe a logic analyzer.

maxgerhardt, 2021-02-07:
I have fully identified the bug. 

The `Adafruit_MCP23017::readINTCAPAB()` method has a bug where it only requests 1 byte for INTCAPA, but before reading INTCAPB it forgets to do another `Wire.requestFrom(MCP23017_ADDRESS | i2caddr, 1);`. Therefore the value for that is always 0xFF.

Will update library.

maxgerhardt, 2021-02-07:
Fixed per 21d60a413ee2cbf2ef061712035421c986ce9fcb, 9c7dc41387ac2a63d133b7c1ed8cf035f632655c and d345231cc9d8e48980bc1129491d065d692052f7, please retest and reopen if necessary.

closed 2021-02-07 by maxgerhardt