maxgerhardt/rotary-encoder-over-mcp23017 #5: how to connect multiple MCP23017 together and operate on one buss

forsakenrecordingstudios, 2020-04-05:
how do i connect multiple MCPs together?

maxgerhardt, 2020-04-06:
> connect multiple MCP23017 together and operate on one buss  
Can you show the exact schematic you're working with? It's hard to guess how you "connect together" the MCP23017. Isn't it the use case shown in the README of this library?

wasimabrar, 2020-08-26:
Can you provide the layout if I wish to connect with  2xMCP23017 or 16 encoders or in general nxMCP23017,where n=<8.?

maxgerhardt, 2020-09-12:
Duplicate the MCP23017 part of the schematics shown in the `README` and but instead of connecting A2,A1,A0 to GND (--> address is <prefix>000)), increment the address so that there is no conflict (e.g., GND, GND, VCC --> Addr 1). 

If you want to work in interrupt mode instead of pulling mode you will have to logically OR the interrupt pins of the devices together, through a OR gate in some logic technology (diode-OR, CMOS OR chip, ...)

maxgerhardt, 2021-02-07:
Closing due to given explanation.

closed 2021-02-07 by maxgerhardt