maxgerhardt/rotary-encoder-over-mcp23017 #6: Arduino IDE

forsakenrecordingstudios, 2020-04-05:
Hello,

   How do i use this library inside Arduino IDE?

maxgerhardt, 2020-04-06:
I **highly** recommend its usage within PlatformIO (https://platformio.org/). 

Installation within Arduino IDE is still possible however. Go into your Arduino IDE library folder (refer to [this reference](https://www.arduino.cc/en/Guide/Libraries)), remove any previous version of the libraries listed in https://github.com/maxgerhardt/rotary-encoder-over-mcp23017/tree/master/lib and copy the version from the repository into the libraries folder. 

You can then start a new sketch to which you must add 2 files: The header of this library ( https://github.com/maxgerhardt/rotary-encoder-over-mcp23017/tree/master/src) and an example file (one of the 2 examples).

maxgerhardt, 2021-02-07:
CLosing due to inactivity and given explanation above.

closed 2021-02-07 by maxgerhardt