maxgerhardt/rotary-encoder-over-mcp23017 #7: [QUESTION] compatibility to NodeMCU?

othmar52, 2020-07-04:
Thanks for sharing your code It's working perfectly on an Arduino Uno.  

Now i am trying to figure out if this is compatible to a [`Node MCU ESP8266`](https://smarthome-blogger.de/guide/nodemcu-esp8266-pinout/#attachment_2055) which has only one Analog pin `A0`.  
According to your circuit the [physical pins](https://miro.medium.com/max/1400/1*qTKdbdMOYuwAgs65WymvIg.png) `12` and `13` of the MCP23017 should be connected to `A4` and `A5`.  

Do you have any advice how to connect and are there any `pinMode(???)` calls necessary?  

Sorry for my noob question but the whole Arduino stuff is rocket science for me...

othmar52, 2020-09-11:
@maxgerhardt bump
sorry for beeing tedious!

maxgerhardt, 2020-09-11:
>According to your circuit the physical pins 12 and 13 of the MCP23017 should be connected to A5 (serial clock) and A4 (serial data).
Since the MCP23017 is an I2C device you need to connect to the I2C pins of your device. A4 & A5 are the ones for the Uno / ATMega328P. In your linked documentation you can see that the I2C pins are on D1 and D2 (SCL, SDA) for the NodeMCU, so you have to use those. No analog pin pinvovled. 

To be 100% sure you can also use this code as the first line in `setup()` to setup the I2C pins on GPIO5 and GPIO4 (equal D1, D2)

```cpp
Wire.begin(4,5); 
```
**this should however be the default already**. (https://github.com/claws/BH1750/issues/12)

othmar52, 2020-09-12:
Thanks a lot for answering my question!!!
Now it's some kind of working.

I've tested with a `KY-040` encoder:
Capturing switch events works perfectly  
Capturing rotation events works 90% of the time. The other 10% shows invalid direction or a rotation event of not connected encoders (invalid pin index). (in my testing with an Arduino UNO it worked 100%)

Testing [with the encoders i want to use in my project](https://www.digikey.at/product-detail/de/bourns-inc/PEC16-4015F-S0024/PEC16-4015F-S0024-ND/3780222) rotation events are not beeing captured at all.

**OFF-TOPIC question:**  
Do you think `10 kOhm` resistors can be the reason for the events not beeing captured?
My circuit for each encoder looks like [this](http://community.axoloti.com/uploads/default/original/2X/a/a12d952c2153928d8ae46598399c943076c192dd.png)

maxgerhardt, 2020-09-12:
The 10kOhm pullups to +3.3V are okay. Can you post what the serial output looks like and what sketch you're using exactly?

Also you should try testing without a breadboard and direct connections to the board instead because there might be crosstalk issues?

othmar52, 2020-09-13:
The sketch i am using:  
https://github.com/othmar52/ui24r-arduceptor/blob/encoder-test/ui24r-arduceptor.ino  
The serial output with this sketch:  
https://github.com/othmar52/ui24r-arduceptor/blob/encoder-test/temp-serial-output.md  

Thanks for the hint that the breadboard can be the reason. i will try a soldered version today

othmar52, 2020-09-13:
I am some kind of a donkey because my encoder circuit's had been wrong:  
As i wasn't able to find any pin description for my encoders i assumed it will be like the numerous schematics i've found in several tutorials

My assumption for pins of [Bourns Inc. PEC16-4015F-S0024 Encoder](https://www.digikey.at/product-detail/de/bourns-inc/PEC16-4015F-S0024/PEC16-4015F-S0024-ND/3780222)
```
      My assumption:                 In fact its like that:
     
     OUTA   GND    OUTB                GND    OUTA   OUTB
      |      |      |                   |      |      |   
    --------------------             --------------------
    |                  |             |                  |
    |                  |             |                  |
    |     ENCODER      |             |     ENCODER      |
    |       TOP        |             |       TOP        |
    |      VIEW        |             |      VIEW        |
    |                  |             |                  |
    |                  |             |                  |
    --------------------             --------------------
          |      |                         |      |
         GND    SW                        GND    SW
```
`KY-040 Encoder` still fires some invalid pin index events - especially on fast rotations.  
But i was not able to reproduce any invalid pin index event with the `Bourns Inc Encoders`.  

So everything seems to work as expected and i can start to build my project.  

@maxgerhardt  **Thank you so much for your help!!!**

closed 2020-09-13 by othmar52