maxgerhardt/rotary-encoder-over-mcp23017 #8: Weird behaviour when using pin 6 and 9 on MCP23017

bartvanheckegithub, 2021-09-10:
Dear Maxgerhardt,

I am trying to use your rotaryEncOverMCP library. It works fine ! Only, when I add a rotaryencoverMCP object in the array with pins 6 and 9 , the ESP32 crashes. This is very strange behaviour. Even without hardware attached. The library doesn't like the numbers 6 and 9...

Could you imagine an explanation for this?

maxgerhardt, 2021-09-10:
Please show the full code (or project with `platformio.ini`) you're having problems with.

bartvanheckegithub, 2021-09-10:
In fact it is this line
`RotaryEncOverMCP rotaryEncoders[] = {
      
       RotaryEncOverMCP(&mcpDraaiKnoppen2, 6, 9,  &RotaryEncoderChanged, 10)

};`

The Esp32 crashes when I add an encoder on pin 6. I thought it was hardware related but it crashes without hardware attached and it's the same for every MCP23017 in the citcuits. I heve got 3 of them. 
Every time I try to add an encoder on an mcp on pin 6 (and 9) it crashes. All the other pins works perfect.

maxgerhardt, 2021-09-10:
And that is with the [PollingExample](https://github.com/maxgerhardt/rotary-encoder-over-mcp23017/blob/master/examples/Polling/PollingExample.cpp)? What's the definition of `mcpDraaiKnoppen2`? Are there other code changes?

bartvanheckegithub, 2021-09-10:
So I am working on a sequencer project. The pcb are soldered so I can't change no connections of the MCP's. I connected all clock lines to portB and all data lines to portA of the MCP's.
I tried the interrupt and the polling code. The same issue. 
Today I copied the example polling code and initialised an encoder on pin 6 and 9 , I upload it to ESP32 and it crashes.
Always these pins . Any other pins works fine!
It sounds crazy but that's the reason I contaced you. I just do not understand it. 
Draaiknoppen2 is just a name (in flemmish) of an encoder, sorry.

So the issue also occurs using the examlple code ...
thank you

maxgerhardt, 2021-09-10:
I see, if the issue occurs with the example code I should be able to reproduce it with an ESP32 and an MCP23017 in a breadboard.

bartvanheckegithub, 2021-09-10:
I guess, but even without any hardware attached, it crashes with these pins after uploading the code. know this sounds crazy...

maxgerhardt, 2021-09-10:
Have you done a sanity check with just a blinky code? How do you observe the crashing? Is there a backtrace? Have you decoded that with `monitor_filters = esp32_exception_decoder` ([docs](https://docs.platformio.org/en/latest/core/userguide/device/cmd_monitor.html#cmd-device-monitor-filters))?

bartvanheckegithub, 2021-09-10:
This is what I get after uploading 
[image]

bartvanheckegithub, 2021-09-10:
I haven't done anything else yet. I'll check out monitor_filters = esp32_exception_decoder .

maxgerhardt, 2021-09-10:
The exception decoder will only be able to decode a human-readable exception text thrown by Arduino-ESP32 or ESP-IDF. That up there looks like nearly completely random data, as if something was interfering with the UART pins -- the exception decoder won't help there. 

I've checked the pinout and GPIO6 and GPIO9 on an ESP32 shouldn't be anything special, the main UART (U0 TXD, U0 RXD) is on GPIO1 and GPIO3. 6 and 9 are SPI signals. (Though if those go to the *internal* flash chip where the program is stored, it will screw things up, not entirely sure about that now)

bartvanheckegithub, 2021-09-10:
But 6 and 9 are pins on the MCP , aren't they? Pin 6 on the esp32 is indeed for spi flash and can't be used.

maxgerhardt, 2021-09-10:
My bad, yes of course they are. Are you using the standard I2C pins too in the ESP32? (SCL=22, SDA=21)?

bartvanheckegithub, 2021-09-10:
Yes I do.

bartvanheckegithub, 2021-09-10:
I have not the intention to rush you but I wondered if you encountered the same issue...

maxgerhardt, 2021-09-10:
I'll be building up the reproduction circuit shortly. 

But I can already say that works with my dev board + minimal circuit, the error is either in code not shown here, or in the hardware design, or in the assembled hardware (which due to placement or soldering errors might be different from the design), and we'd need to see schematics and the full code to say anything more about the issue. But one step at a time.

bartvanheckegithub, 2021-09-10:
Ok, nice.
One thing more, when I upload the next code to my esp32 (without anything attached) it crashes (see picture of serial monitor I send)`. In fact this is your example code. When I change 6 to 7 , it works !!

`#include <Arduino.h>
#include <Wire.h>
#include <Adafruit_MCP23017.h>
#include <Rotary.h>
#include <RotaryEncOverMCP.h>

/* Our I2C MCP23017 GPIO expanders */
Adafruit_MCP23017 mcp;

//Array of pointers of all MCPs if there is more than one
Adafruit_MCP23017* allMCPs[] = { &mcp };
constexpr int numMCPs = (int)(sizeof(allMCPs) / sizeof(*allMCPs));

/* function prototypes */
void RotaryEncoderChanged(bool clockwise, int id);

/* Array of all rotary encoders and their pins */
RotaryEncOverMCP rotaryEncoders[] = {
        // outputA,B on GPA7,GPA6, register with callback and ID=1
        RotaryEncOverMCP(&mcp, 6,9, &RotaryEncoderChanged, 1)
};
constexpr int numEncoders = (int)(sizeof(rotaryEncoders) / sizeof(*rotaryEncoders));

void RotaryEncoderChanged(bool clockwise, int id) {
    Serial.println("Encoder " + String(id) + ": "
            + (clockwise ? String("clockwise") : String("counter-clock-wise")));
}

void setup(){
    Serial.begin(9800);
    Serial.println("MCP23007 Polling Test");

    mcp.begin();      // use default address 0

    //Initialize input encoders (pin mode, interrupt)
    for(int i=0; i < numEncoders; i++) {
        rotaryEncoders[i].init();
    }
}

void pollAll() {
    //We could also call ".poll()" on each encoder,
    //however that will cause an I2C transfer
    //for every encoder.
    //It's faster to only go through each MCP object,
    //read it, and then feed it into the encoder as input.
    for(int j = 0; j < numMCPs; j++) {
        uint16_t gpioAB = allMCPs[j]->readGPIOAB();
        for (int i=0; i < numEncoders; i++) {
            //only feed this in the encoder if this
            //is coming from the correct MCP
            if(rotaryEncoders[i].getMCP() == allMCPs[j])
                rotaryEncoders[i].feedInput(gpioAB);
        }
    }
}

void loop() {
    pollAll();
}`

maxgerhardt, 2021-09-11:
I can indeed verify an eternal bootloop with the code you've posted. Together with the exception decoder at 115200 baud I get 

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40080fa5  PS      : 0x00060330  A0      : 0x800d10f3  A1      : 0x3ffe3b50
A2      : 0x00000009  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x3f400a60  A9      : 0xffffffff
A10     : 0x0f000000  A11     : 0x00000054  A12     : 0x00000001  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0f000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x40080fa5:0x3ffe3b50 0x400d10f0:0x3ffe3b70 0x400d0ecd:0x3ffe3b90 0x400e1fef:0x3ffe3bb0 0x40081fd1:0x3ffe3bd0 0x400821d1:0x3ffe3c20 0x40079247:0x3ffe3c40 0x400792ad:0x3ffe3c70 0x400792b8:0x3ffe3ca0 0x40079465:0x3ffe3cc0 0x400806da:0x3ffe3df0 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20
  #0  0x40080fa5:0x3ffe3b50 in __pinMode at C:\Users\Max\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-gpio.c:115
  #1  0x400d10f0:0x3ffe3b70 in Rotary::Rotary(char, char) at lib\Rotary-master/Rotary.cpp:88
  #2  0x400d0ecd:0x3ffe3b90 in RotaryEncOverMCP::RotaryEncOverMCP(Adafruit_MCP23017*, unsigned char, unsigned char, void (*)(bool, int), int) at src/RotaryEncOverMCP.h:30
      (inlined by) __static_initialization_and_destruction_0 at src/main.cpp:21
      (inlined by) _GLOBAL__sub_I_mcp at src/main.cpp:60
  #3  0x400e1fef:0x3ffe3bb0 in do_global_ctors at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/cpu_start.c:510 (discriminator 3)
  #4  0x40081fd1:0x3ffe3bd0 in start_cpu0_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/cpu_start.c:396
  #5  0x400821d1:0x3ffe3c20 in call_start_cpu0 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/cpu_start.c:262
  #6  0x40079247:0x3ffe3c40 in ?? ??:0
  #7  0x400792ad:0x3ffe3c70 in ?? ??:0
  #8  0x400792b8:0x3ffe3ca0 in ?? ??:0
  #9  0x40079465:0x3ffe3cc0 in ?? ??:0
  #10 0x400806da:0x3ffe3df0 in ?? ??:0
  #11 0x40007c31:0x3ffe3eb0 in ?? ??:0
  #12 0x4000073d:0x3ffe3f20 in ?? ??:0
```

So the core really doesn't like it that `pinMode` is being called in a constructor before the framework gets a chance to initialize. Let me try to fix that by refactoring.

maxgerhardt, 2021-09-11:
These bootloader messages appear at 115200 baud, so with your `Serial.begin(9800);` and the serial monitor being set to this, you will have seen that garbage UART output.

Also 9800 is a really weird baud rate, 9600 is the next standard one. I'd advise to use a high baud-rate since in the polling mode, printing at low baud rates slows the program down and it eventually misses rotations.

maxgerhardt, 2021-09-11:
Oh yeah, that's a.. major bug that went on unnoticed.. Commit https://github.com/maxgerhardt/rotary-encoder-over-mcp23017/commit/1699a61b68c2da5667a394357467b778fea39044 fixes a flaw that when the `RotaryEncOverMCP` object is intitialized it initializes a `Rotary` object which uses the given pin number infos (which are supposed to represent pin numberings on the MCP23017) initializes these pin numbers locally with `pinMode()` and `digitalWrite()` pullup etc. 

Calling `pinMode()` in a constructor before the framework has been initialized leads to a crash in Arduino-ESP32, and also it is of course wrong to initialize these pins locally when they're supposed to only be referencing the pin number on the MCP23017. The commit fixes that.

Please retry your example against the latest code found in this repository. And I'd also recommend to change the baud rate in the code to 115200, since then it's fast and in sync with bootlaoder / crash messages.

I'm getting on my hardware setup now

```
--- Miniterm on COM4  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
MCP23007 Polling Test
Encoder 1: clockwise
Encoder 1: clockwise
Encoder 1: clockwise
Encoder 1: counter-clock-wise
Encoder 1: counter-clock-wise
Encoder 1: counter-clock-wise
Encoder 1: counter-clock-wise
```

with the circuit described in the README.

[image]

bartvanheckegithub, 2021-09-11:
Waaw, it works , wonderfull !! Great moment, good thinking !
It had to be in software...
For a week I thought I was going nuts ;) 
Thank you Max, nice to meet you, keep up the good work.

bartvanheckegithub, 2021-09-11:
By the way , I changed my baud Rate to 9600. When I put 115200 it doesn't print in the Serial monitor !

bartvanheckegithub, 2021-09-11:
Nice to know about exception decoder too, I didn't know it. Seems to be very helpful.
[image]

A bad quality picture of my self-assembled sequencer hardware. These encoders are gonna turn !!

maxgerhardt, 2021-09-11:
>By the way , I changed my baud Rate to 9600. When I put 115200 it doesn't print in the Serial monitor !

Make double sure the [`monitor_speed`](https://docs.platformio.org/en/latest/projectconf/section_env_monitor.html#monitor-speed) is set to the same baud rate as your code is using, otherwise that won't work. I had no problems at 115200 baud (and it also works for me at a clean 9600 baud, even if 115200-baud bootloader messages trash the output at the beginning)

```
--- Miniterm on COM4  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
��)�
���B�ަ����1!y��!���!�MCP23007 Polling Test
Encoder 1: clockwise
Encoder 1: clockwise
Encoder 1: clockwise
Encoder 1: counter-clock-wise
```

maxgerhardt, 2021-09-11:
I'll close this issue as solved then. I've also update the standard interrupt example to work on the ESP32 and added 2 new interrupt-based examples that do processing on a separate CPU core and additionally do buffering so that the main task can access the buffered data comfortably: https://github.com/maxgerhardt/rotary-encoder-over-mcp23017/tree/master/examples

closed 2021-09-11 by maxgerhardt

bartvanheckegithub, 2021-09-11:
Nice work, thanks.