maxgerhardt/rotary-encoder-over-mcp23017 #14: Working on polling but not Interrupt examples

stanleyseow, 2023-06-18:
Hi, 

I have setup a rot encoder at mcp23017 pins 0, 1 ...

It is working ( a bit delay response ) on the **polling example** but totally no response on the **Interrupt examples** on D3 on the Arduino Nano

I didn't change anything in the example code except the below...

I've used Adafruit mcp23017 before, so I knew the mcp pin numbering goes from 0 to 15

`
    RotaryEncOverMCP(&mcp, 0, 1, &RotaryEncoderChanged, 1),

`

maxgerhardt, 2023-06-18:
And you did connect either INTA or INTB to the D3 pin?

[image]

stanleyseow, 2023-06-19:
Yes, I did connect INTA to D3 ...

maxgerhardt, 2023-06-19:
Please show a clear photo of your setup / wiring.