maxgerhardt/rotary-encoder-over-mcp23017 #15: Problem to install

KinoLab07, 2023-08-15:
Hi, I'm quite new in arduino... I'm on OsX system and when i try to install the library in the standard way (directly with the .zip file) I receive this error:
[image]

So, there is a particular way for install this library... 

Thx for your works

maxgerhardt, 2023-08-15:
I've never in my life seen PlatformIO throw such an error message. Is that the Arduino IDE 2.0 or PlatformIO? Because this repository is **not** a Arduino IDE library, it's a PlatformIO firmware project. To install it in the Arduino IDE, you have to copy the library folders from https://github.com/maxgerhardt/rotary-encoder-over-mcp23017/tree/master/lib in `<User folder>\Arduino\libraries`, then create a new sketch and save it, then in the same folder as the sketch (.ino) file, add the https://github.com/maxgerhardt/rotary-encoder-over-mcp23017/blob/master/src/RotaryEncOverMCP.h file.

KinoLab07, 2023-08-15:
Ok... Thk...
Now how can I try with one encoder..
When I open the empty scketch that I've created, it open the .h file too... but i don't know how test if the encoder works as midi... Sorry but I'm newer and learning with the Arduino

maxgerhardt, 2023-08-15:
The one MCP23017 chip + one rotary encoder circuit is shown in https://github.com/maxgerhardt/rotary-encoder-over-mcp23017#circuits.

That hardware setup also means you can use https://github.com/maxgerhardt/rotary-encoder-over-mcp23017/blob/master/examples/Polling/PollingExample.cpp without any adaptations. It just declares one rotary encoder on the chip's GPA7, GPA6 pins.

Also note that you can look into the datasheet (https://ww1.microchip.com/downloads/en/devicedoc/20001952c.pdf) at any time to see the pinout of the chip. But you might not have the bare chip and a module with the chip instead where all the pins are labeled.

KinoLab07, 2023-08-15:
okkkk... I've copy the pollingExample.cpp in the same directory of the .ino, and when I try to verify I receive this error
[image]

KinoLab07, 2023-08-15:
Yes is inside the folder
[image]

maxgerhardt, 2023-08-15:
Oh? Weird. Well first of all, having both the `PollingExample.cpp` and the `.ino` file is very weird, you should delte one of them and put the content of https://github.com/maxgerhardt/rotary-encoder-over-mcp23017/blob/master/examples/Polling/PollingExample.cpp in the one remaining file.

Does the error change when you use
```
#include "RotaryEncOverMCP.h"
```
instead?

maxgerhardt, 2023-08-15:
The `PollingExample` sketch is just a very basic test. There isn't anything in that code to interact with anything MIDI. You would have to combine the codes as needed.

maxgerhardt, 2023-08-15:
In any case, since you see that the rotary encoder is working in the serial monitor with the test sketch, this library's purpose is fullfilled and there's no issue with it.

closed 2023-08-15 by maxgerhardt