maxgerhardt/rotary-encoder-over-mcp23017 #16: Example with multiple MCP23017 please?

andrewerrington, 2023-10-21:
Hello. Can you post an example (or link to an existing one) showing how to use the library with several port expanders please?
I have a project to use three port expanders with six encoders on each (total 18 encoders) and I would like to know how easy it's going to be.
Thank you.

maxgerhardt, 2023-10-22:
See https://github.com/maxgerhardt/rotary-encoder-over-mcp23017/blob/master/examples/Polling/PollingExample.cpp but create as many `Adafruit_MCP23017` objects as you need, reference them in the `allMCPs` array as well as `rotaryEncoders` for your each of your 18 encoders (with their port pins) and in `setup()`, `.begin()` on each `Adafruit_MCP23017` with the address you gave them (that is, the number described by the address pins (A2, A1, A0) with GND=0, VCC=1).

andrewerrington, 2023-10-22:
Ah. Got it. Thanks!

andrewerrington, 2023-10-22:
Ah. Got it. Thanks!

closed 2023-10-22 by andrewerrington